freifunk/meshkit-firmware #27: MQTT: solarfred - mosquitto Kopplung

FreifunkUFO, 2015-01-09:
Für das Einsammeln der Solardaten, Temperaturfühler (und mehr) sollten wir MQTT verwenden.
http://mosquitto.org/2011/08/mosquitto-on-openwrt/

mmunz, 2015-08-01:
valid for leipzig firmware only

FreifunkUFO, 2015-08-27:
betrifft nicht leipzig, sondern alle mit Interesse an Heimautomatisierung..

mosquito ist im leipziger juni-build (trunk) mit dabei, ticket kann geschlossen werden.

root@113-32:~# opkg list | grep mosquitto
libmosquitto - 1.4.2-2 - Library required for mosquitto's command line client tools, also for use by any third party software that wants to communicate with a mosquitto server.  Should be useable for communicating with any MQTT v3.1/3.1.1 compatible server, such as IBM's RSMB, in addition to Mosquitto This package is built with SSL support
libmosquitto-nossl - 1.4.2-2 - Library required for mosquitto's command line client tools, also for use by any third party software that wants to communicate with a mosquitto server.  Should be useable for communicating with any MQTT v3.1/3.1.1 compatible server, such as IBM's RSMB, in addition to Mosquitto This package is built without SSL support
lua-mosquitto - 0.1-1 - Lua bindings to libmosquitto
mosquitto - 1.4.2-2 - Mosquitto is an open source (BSD licensed) message broker that implements the MQTT protocol version 3.1 and 3.1.1. MQTT provides a lightweight method of carrying out messaging using a publish/subscribe model.  This package also includes some basic support for configuring via UCI This package is built with SSL support
mosquitto-client - 1.4.2-2 - Command line client tools for publishing messages to MQTT servers and subscribing to topics. This package is built with SSL support
mosquitto-client-nossl - 1.4.2-2 - Command line client tools for publishing messages to MQTT servers and subscribing to topics. This package is built without SSL support
mosquitto-nossl - 1.4.2-2 - Mosquitto is an open source (BSD licensed) message broker that implements the MQTT protocol version 3.1 and 3.1.1. MQTT provides a lightweight method of carrying out messaging using a publish/subscribe model.  This package also includes some basic support for configuring via UCI This package is built WITHOUT SSL support.

closed 2015-08-27 by FreifunkUFO